rsta2/minisynth #2: Default config seems to be for a multicore pi 1

Baggypants, 2020-02-27:
`configure` seems to write 

`echo "DEFINE += -DARM_ALLOW_MULTI_CORE -DSAVE_VFP_REGS_ON_IRQ" >> Config.mk`

even if you pass 1 for the rpi version. This leads to a compilation error later on.

```
In file included from logger.cpp:24:
../include/circle/multicore.h: In static member function 'static unsigned int CMultiCoreSupport::ThisCore()':
../include/circle/multicore.h:71:20: error: 'CORES' was not declared in this scope
   71 |   return nMPIDR & (CORES-1);
      |                    ^~~~~
```

removing `-DARM_ALLOW_MULTI_CORE` allows the compilation to continue.

Baggypants, 2020-02-27:
Oh I guess it would help if I checked out a the v2.0 tag

rsta2, 2020-02-28:
This was an issue in the latest development version. I've corrected it.

Baggypants, 2020-02-29:
Seems to be fixed, but I can't clean compile because of the issues I'm having in #3

Baggypants, 2020-02-29:
Confirmed fixed.

rsta2, 2020-03-01:
Thanks. Because the issue was in a development version, we do not need to wait for the next release and can close this issue.

closed 2020-03-01 by rsta2