rsta2/minisynth #3: missing math.h

Baggypants, 2020-02-27:
using tag v2.0 and head.

```
  CPP   minisynth.o
In file included from minisynth.cpp:21:
math.h:23:10: fatal error: math.h: No such file or directory
   23 | #include <math.h>
      |          ^~~~~~~~
```

I'm assuming from having a look in circle that you used [circle-stdlib](https://github.com/smuehlst/circle-stdlib) somehow? It's not obvious to me how to include it.

Baggypants, 2020-02-28:
Tag v1.0 compiles and runs.

rsta2, 2020-02-28:
<math.h> must be provided by the toolchain. Which one are you using?

Baggypants, 2020-02-28:
Ok, I'm just using Fedora packaged `arm-none-eabi-gcc-cs-c++-1:9.2.0-1.fc31.x86_64` which does provide math.h in [/usr/arm-none-eabi/include/c++/9.2.0/math.h](https://koji.fedoraproject.org/koji/fileinfo?rpmID=18788479&filename=/usr/arm-none-eabi/include/c%2B%2B/9.2.0/math.h)

I tried stuffing `-I /usr/arm-none-eabi/include` in to https://github.com/rsta2/circle/blob/d2cc5335229a7ecb889ecd6a8d93a5acde79426e/Rules.mk#L125 ,which I admit was a bit of a punt, but it didn't seem to like it. You can freely chuckle at my ineptitude.

Baggypants, 2020-02-28:
Oh wait, I've just realised there's one in 
```
/arm-none-eabi-gcc-cs-c++-1:9.2.0-1.fc31.x86_64
/usr/arm-none-eabi/include/c++/9.2.0/math.h` 
```
and another in 
```
arm-none-eabi-newlib-3.1.0-4.fc31.noarch
/usr/arm-none-eabi/include/math.h
```
I'll have to check which one I thought I was including this evening.

rsta2, 2020-02-28:
Normally the system header files will be found automatically by the compiler, if the option `-nostdinc` is not specified, what is the case. I'm using the toolchain from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads) (gcc-arm-9.2-2019.12-x86_64-arm-none-eabi.tar.xz). It should work in any case with this one.

rsta2, 2020-02-29:
There is a new branch "test" in the repo. I reverted the commit, which moved the math functions over to the toolchain libraries there and implemented the fabsf() and sqrtf() functions, so that they work with AArch64. I successfully built this version for RPi 1 on Raspbian Buster on a RPi 3B and it runs. It should build on a RPi 4 too.

Baggypants, 2020-02-29:
Oh, well, I just got the toolchain stuff sorted too! It compiles and runs cleanly in master.

rsta2, 2020-03-01:
No problem! It's good, that it builds now with the recommended toolchain for you. I can easily remove the "test" branch again and it wasn't a big effort.

rsta2, 2020-03-01:
I have removed the "test" branch. I think, this can be closed.

closed 2020-03-01 by rsta2