rsta2/minisynth #4: Last note priority polyphony.

Baggypants, 2020-02-28:
Switch from first note priority to last note priority for polyphony. i.e. when the 5 note is played during 4 already sounding notes(rpi 1), the first is cut short and the new note is sounded.

rsta2, 2020-02-28:
Is this the way, professional MIDI instruments are working? I'm not a professional musician, that's why I'm asking. Currently the last note is simply ignored, when all voices are active and another note-on arrives. I have to see, how your suggestion can be implemented.

Baggypants, 2020-02-28:
In my experience nearly all are last note priority, however the Roland Juno 106 was first note priority. Most modern synths have an option to choose and I suppose if you implemented it you could have a compile time option to set it either way. 

Or you could just tell people "I was going for that classic Juno workflow" and save yourself some work :smile:

rsta2, 2020-02-29:
The Juno 106 is maybe a little bit to old to act as reference here. ;) So I will try to implement the last note priority polyphony (maybe as an option), when I finished the work on my current project. This may be in a month.

rsta2, 2020-03-09:
It's implemented as an option, which is on by default.

Baggypants, 2020-03-09:
Much easier for me to play, I can now be certain than new notes will sound when I press the key. Thanks so much!

rsta2, 2020-03-10:
Glad to hear that. You are welcome. I think, this can be closed.

closed 2020-03-10 by rsta2