rsta2/minisynth #5: Add support for midi cc

Baggypants, 2020-02-28:
MIDI cc controllers to manage volume, filter cutoff & resonance and program change would mean the synth could work more flexibly headless.

rsta2, 2020-02-28:
This is interesting, but it will be difficult for me to implement it, because I do not have a MIDI CC controller for testing. Perhaps it's an idea for someone, who owns a MIDI CC controller and is able to program it.

VasilyMedved, 2020-03-17:
I guess cc command can be emulated:
http://www.csounds.com/maldonado/vmci.html
https://vmpk.sourceforge.io/

In my opinion support of midi commands to "filter cutoff & resonance and program change" as Baggypants suggest is crucial for such project as this, almost as crucial as ability to send notes and hear sounds from synth )

rsta2, 2020-03-18:
I understand that. I will have a look at it. To be prepared the goal would be to support the following MIDI messages, right?

```
Bn 47 vv	Filter resonance
Bn 4A vv	Filter cutoff
Cn vv		Program change

n		MIDI channel
vv		Value

(all numbers in hex)
```

VasilyMedved, 2020-03-18:
Thank you for quick responce @rsta2 !
I guess the good point to start is Filter controls and ADSR controls.
In my opinion, ideally all controls could be controlled via midi CC (except HELP obviously)
I have hardware midi controller (AKAI MPK mini, quite popular one) and PI, so will be happy to test this feature asap.

rsta2, 2020-03-19:
@VasilyMedved Thanks for the testing offer! I will come back on it. I think, I will start with the filter controls. Unfortunately currently I'm working on the Circle libraries again, which are the basis for MiniSynth Pi. I think, I will work on MiniSynth in up to two weeks again.

Baggypants, 2020-03-21:
I also have a couple of programmable hardware controllers I can test with.

rsta2, 2020-03-21:
@Baggypants I will come back on it. Thanks!

rsta2, 2020-03-27:
@Baggypants @VasilyMedved I have added support for MIDI CC for the filter and for program change too. I come back on your offer to test and will be happy about feedback.

Baggypants, 2020-03-28:
Works a treat. Program change, filter on cc 74 and resonance on cc 71.  Champion job!

rsta2, 2020-03-29:
Great that it works! Thanks! Is there something more (more parameters to control) I could do about MIDI CC with MiniSynth or is this enough for the moment?

Baggypants, 2020-03-29:
Volume to midi cc 7 would be really helpful. Now I can see what the changes need to be I’ll have a go at mapping some more myself and make a pull request.

rsta2, 2020-03-29:
OK, I have added MIDI CC 7. A useful MIDI CC mapping would be great, if you want to deal with this.

VasilyMedved, 2020-04-05:
Hi!
Finally got chance to test)
I was able to send midi CC commands and hear feedback.
However, changing some parameters and playing notes simultaneously causes sound glitches and jumps in parameter value.
Looks like application doesn't like simultaneous Midi CC and Midi Notes

rsta2, 2020-04-06:
Yes, there is no smooth transition for the MIDI CC parameters, which may cause sound glitches. I currently do not know, how to realize an test this. Maybe too much for this project. Thanks for testing.

romix123, 2022-06-04:
Does MiniSynth now actually respond to CC and PC messages. I run the current build (v4.1). MiniSynth does not seem to respond to my custom controller (which delivers the correct CC and PC messages according to MidiView). My midi-cc.txt file contains the codes my controller submits.
It also does not respond to my Arturia MinLab v2, btw.

rsta2, 2022-06-05:
Yes, this should work. Can you post your *midi-cc.txt* file here?

romix123, 2022-06-06:
I can get Minisynth to recognise the Arturia Minilab allright now, still no luck with my SoundCore Control unit (SAMD21 based design of my own, see https://github.com/romix123/SoundCoreControl) 
This is my midi-cc.txt
#
# MIDI CC Mapping SoundCore Control
#
LFOVCOWaveform=94
LFOVCOFrequency=74
VCOWaveform=
VCOModulationVolume=
VCODetune=
LFOVCFWaveform=
LFOVCFFrequency=
VCFCutoffFrequency=
VCFResonance=71
EGVCFAttack=73 
EGVCFDecay=80
EGVCFSustain=64
EGVCFRelease=69
VCFModulationVolume=
LFOVCAWaveform=
LFOVCAFrequency=
EGVCAAttack=18
EGVCADecay=19
EGVCASustain=20
EGVCARelease=21
VCAModulationVolume=
ReverbDecay=
ReverbVolume=91
SynthVolume=07

This is what SoundCore Control currently produces:
Program change 0,1, ,C0 00 
Program change 1,1, ,C0 01 
Program change 2,1, ,C0 02 
Program change 3,1, ,C0 03 
Program change 4,1, ,C0 04 
Program change 5,1, ,C0 05 
Controller Sound Attack Time: 6,1,6,                                   B0 49 06 
Controller General Purpose Button 1 (on/off): 54,1,54,   B0 50 36 
Controller Hold Pedal (on/off): 57,1,57,                               B0 40 39 
Controller Hold 2 Pedal (on/off): 114,1,114,                        B0 45 72 
Controller Celeste Level: 102,1,102,                                    B0 5E 66 
Controller Sound Timbre: 32,1,32,                                      B0 47 20 
Controller Sound Brightness: 111,1,111,                              B0 4A 6F 
Controller Volume (coarse): 50,1,50,                                 B0 07 32 

The Hex CCs correspond with decimals in the midi-cc file (the midi file contains more entries, I have only mapped the EGVC pARAMS, WAVEFORM, FREQ and SynthVolume). No response to any of the CC and PC messages. My guess is that the device does not properly;y enumerate in MiniSynth. I haven't figured out how to test this.

romix123, 2022-06-07:
Thanks. SoundCore Control does not enumerate. I think it is an issue with the USBMIDI arduino lib. My Nektar keyboard does enumerate.
It is a pity Minisynth only supports one usb device (the whole purpose of making SoundCore Control was to allow for headless minimal synths. There also is a SoundCore with an ESP and DAC as a standalone synth running Marcel Licence’s code).

rsta2, 2022-06-07:
I have added "Support multiple USB MIDI controllers" to the "Hardware support" project Todo list.

romix123, 2022-06-07:
Thanks

Verstuurd vanaf mijn iPad

Verstuurd vanaf mijn iPad

rsta2, 2022-07-09:
The support for multiple USB MIDI controllers (up to two) has been implemented on the *master* branch.

closed 2023-11-10 by rsta2