rsta2/minisynth #6: Randomly freezes

Baggypants, 2020-02-28:
Occasionally the synth sounds continuously and the screen becomes unresponsive on my pi 1. I’m using the official psu for the pi1.

rsta2, 2020-02-28:
I have to try, if I can reproduce this. Are you using an USB MIDI controller or serial MIDI?

Baggypants, 2020-02-28:
I'm using usb controllers. One powered by the pi and one externally powered (not both at the same time though).

Baggypants, 2020-02-28:
Also I was working from the v1.0 tag so if I get HEAD compiled it might not be a problem anyway.

rsta2, 2020-02-29:
Unfortunately I wasn't able to reproduce this issue so far. I tested v1.0 on a RPi 1 B+, compiled with GCC 9.2.1. Another question: Did you use I2S or the normal headphone jack and are there any things I could do to trigger this problem (e.g. specific mouse actions)?

Baggypants, 2020-02-29:
I’m on the original Pi1 using the headphone jack. It doesn’t do it often but it’s been related to playing very quickly. I’ll try and get something reproduceable.

Baggypants, 2020-02-29:
I’ve also failed at setting up cross compiling using the toolkit you suggested which is why it’s taking me so long. I’m going to try compiling the latest version on my pi4 for the pi1.

rsta2, 2020-02-29:
I'm afraid, compiling on the RPi 4 won't work too. The source needs a compatible math library to build. Earlier I had implemented the math functions by myself, but in [this commit](https://github.com/rsta2/minisynth/commit/449d60050b5b6ae994e9b4e14ff8134d0676e0fc) I moved over to the library provided by the toolchain, which should provide faster functions. Perhaps I have to take back this change to be able to build the project on more platforms. The toolchain I've referenced can build the project, but if it doesn't work for you, it does not help.

Baggypants, 2020-02-29:
Now I've compiled the newer version I can't get it to freeze.

rsta2, 2020-03-01:
That's good news! :) This may also be caused by a new Circle submodule included in the newer version of MiniSynth Pi. I think, we keep this open for a while, if some similar problem occurs somewhere. Thanks for testing!

rsta2, 2021-04-08:
I have noticed a freeze of MiniSynth Pi on the RPi 1 too. This does not seem to happen on the RPi 2-4.

rsta2, 2021-04-11:
How it seems, this issue is difficult to catch. Sometimes the RPi freezes after some minutes, sometimes nothing happens for hours. I wasn't able to find the reason with a JTAG debugger so far. Because it did only happen on the RPi 1 so far, maybe the next release can only be done for the RPi 2-4?

rsta2, 2021-04-23:
I suppose, the bug has been fixed. It was an issue in the Circle framework. Only the RPi 1 was affected. I think, this can be closed.

closed 2021-04-23 by rsta2