rsta2/minisynth #8: LFO Enhancements

Baggypants, 2020-02-29:
The Pitch LFO works as I'd expect it but the Filter and Amplitude LFO's don't.

If I set the filter cutoff to 50% and the LFO depth to 100% I'd expect the cutoff to sweep from 0% to 100%  using the specified wave form at the set frequency. (not necessarily see the slider but the audible effect ). Changing the LFO depth to 50% it would sweep from 25% to 75%, etc., but it seems that increasing the depth just means it tends towards 0% and doesn't move. 

Equally for the amplitude LFO, I'd expect with the Master volume at 50% and LFO set to 100 the effect would be as if the master volume was sliding from 0 to 100%

Finally the LFOs don't go slow enough. For reference the Korg Minilogue has a range of 0.05Hz to 28Hz with 1024 increments. A good compromise might be 0-5Hz in 10 x 0.5Hz steps.

rsta2, 2020-03-01:
Thanks for this valuable information! I will work on this together with the "last note priority polyphony" modification in a month, when I have more time.

rsta2, 2020-03-09:
I finished my other work quicker, so I was able to implement your suggestions already. Now the VCF and VCA LFOs should be much more useful. Thanks again.

Baggypants, 2020-03-09:
Works a real treat!

rsta2, 2020-03-10:
Great! So this issue is done.

closed 2020-03-10 by rsta2