rsta2/minisynth #9: Track Demo

Baggypants, 2020-02-29:
As a thank you here is an instrument demo. The first half is only clean sounds using minisynth. The second half adds drums and some effects.

https://therobinsonfamily.net/nextcloud/index.php/s/dJo5AAprxXgDacz

rsta2, 2020-03-01:
Thank you very much. You awaken the instrument to life. I'm personally better in programming but in making music and I never heard MiniSynth Pi in this way before. As it seems, one can really make music with it!

VasilyMedved, 2020-04-19:
Hi,
I guess its may be interesting to hear more tracks with minisynth :)
Here is mine:

https://soundcloud.com/bear_me/bear-me-vas

All sounds except drums is the subject.  As for effects, its just a bit of reverb and EQ - especially EQ as my Raspberry Pi is very noisy.
Hope you'll enjoy)

rsta2, 2020-04-20:
@VasilyMedved Great track! Thank you.

rsta2, 2023-11-10:
Great video about MiniSynth Pi by Floyd Steinberg:

https://youtu.be/1P6NX9PQgbE

Listen to the improvisation at 6:56:

https://youtu.be/1P6NX9PQgbE?t=416