rsta2/minisynth #10: Audio out via hdmi

Baggypants, 2020-03-29:
I have an original Pi B which doesn’t have the i2s pins exposed. Is it possible to compile minisynth Pi to use HDMI audio?

rsta2, 2020-03-29:
Currently not. I'm also not sure, if the RPi B could do all the calculations and VCHIQ HDMI audio in time. The software architecture for HDMI audio is much different from PWM and I2S audio. HDMI audio needs some Linux kernel driver emulation in Circle and tasks. It's unfortunately not that simple.

Baggypants, 2020-03-29:
Fair enough.

rsta2, 2021-04-01:
There is some beta HDMI audio out support now in MiniSynth Pi. You have to specify the option `--hdmi` in the first place to `configure` before build to enable this support. Unfortunately there is some delay between a note on/off event and the reaction on the HDMI output. This is caused by the hardware and cannot be eliminated.

rsta2, 2021-04-11:
Did anybody test the HDMI sound support already?

Unfortunately there is a problem with the driver on the Raspberry Pi 4. It generally works, but I'm not sure, if it works in any case. Especially with monitors with pixel resolutions of over 1920x1080 there may be problems, because I do not known, how to fetch the right HDMI audio clock rate and pixel clock rate there.

Currently I do not have hope, that I will be able to solve this and I'm thinking about removing the HDMI sound support again from MiniSynth Pi. If there is somebody, which wouldn't be happy about this, please let me know. Currently I think that removing it would not be a big loss, given that there is an audio delay hear-able and playing is not much fun with this.

rsta2, 2021-04-16:
For the reasons, noted in my previous comment, I have removed the HDMI support again from MiniSynth Pi. I suppose, there will not be another attempt to add HDMI support, so this can be closed.

closed 2021-04-16 by rsta2