rsta2/minisynth #11: Patch reading issue

Baggypants, 2020-12-22:
Two issues

1) modifying a patch setting using the mouse seems to cause it to change to another patch

2) if you modify a patch, switch to another patch using midi program change and then switch back with midi program change then the modifications are still there. (which is kind of fine because of the issue above)

rsta2, 2020-12-23:
Can you please explain in more detail the first issue. Which actions you do, what you expect and what happens instead. I cannot reproduce a wrong behavior here.

The second issue seems to be the normal intended behavior. There is always one active patch, which can be selected using the PATCHES tab or MIDI program change. The edit controls on the MAIN tab modify this active patch. The settings for one patch will be kept, if one changes to another active patch and back later. To save the settings for the active patch on the SD card, you have to click the SAVE button, otherwise its settings will get lost on reboot.

Baggypants, 2020-12-27:
I'm going to try again with a wired mouse. I think using a wireless one may be causing other issues.

rsta2, 2021-04-01:
I have found some similar problem with an other USB keyboard attached now. The keyboard sends program change MIDI messages on channel 3 automatically under special circumstances.

To fix this problem, I added some MIDI channel filtering. On the end on *src/config.h* one can select the MIDI channels to be recognized by setting `MIDI_CHANNEL_FILTER` before build. For example when setting it to 0x0001 only the first channel will be recognized, 0x0003 channel 1 and 2. There is one bit for each channel. By default all channels are recognized as before.

I don't know, if this helps to fix your problem?

rsta2, 2023-11-10:
There has been no reaction from the originator of the issue for a long time. The issue cannot be solved without interaction and will be closed.

closed 2023-11-10 by rsta2