rsta2/minisynth #12: Feature Requests

Lemonmeth, 2021-01-31:
Hi there, thanks for a fantastic project, i found this searching for arduino/rpi synth. 
The best this is that it just works, and with a minimal footprint, great!

So a few suggestions when time allows:

glissando
add another osc
midimap on the go

i realize these are quite big but am prepared to donate as much as my pocket allows, even reccuring monthly to allow some workhours on this.
unfortunatley im not a programmer, just a musician.

i can assist in testing atleast...

Once again thanks for this project, and thank you for your time.
Take care

rsta2, 2021-01-31:
Thanks for the feature suggestions. A 3rd oscillator could be added, if two are not enough. I recently added a 2nd one, which can be de-tuned by a semitone +/- from the main oscillator and already provides more rich sounds, thought. I did not create a new release with it so far.

I'm not a musician and not a professional audio programmer. I still have to learn a lot in this area. I guess, glissando enables the smooth transition of the VFO frequency from one note to another. I'm afraid, this is difficult to implement with the current design of MiniSynth Pi. Unfortunately I did not found useful information about midimap on the go so far.

MiniSynth Pi is a hobby project and was created to demonstrate the creation of a audio synthesizer using the Raspberry Pi and my bare metal framework Circle, which is my main project.

Lemonmeth, 2021-01-31:
A third wave osc would be great, for even more versatility. A suggestion there would be to have two tabs above the filter-column in the ui, for vco 1 and 2. glissando/portamento is not a big deal, still would be amazing if possible.

Regarding the midmap, he option to manually map midi-messages etither CC-nr or the hex-value in the code before building would surley suffice, or if possible a textfile directly on the FAT rootpartition. i have a homebuilt midicontroller based on the atmega32u4 which i can reprogram to send other CC:s. 

I fully understand if these request overstretches the purpose of this project, still the simplicity and functionality is on point (and exactly what iv been searching for, unfortunatley im pre-beginner when it comes to C/C++), and again i would happily donate a fair sum to allow time in this.

So if if you have the time and energy, it would be greatly appreciated.
Take care

rsta2, 2021-02-01:
I will see, what I can do. I cannot make promises. At the moment I'm working on my Circle project, so it will take some time. This project is for fun, not to earn money. So thanks for your offer to donate, but it wouldn't help.

Lemonmeth, 2021-02-01:
i fully understand, and whatever you decide - best of luck to your circle project.

Take care

rsta2, 2021-04-02:
I had some time for this project now and implemented the configurable MIDI CC mapping. There is a new config file *midi-cc.txt* in the *config/* directory, which can be edited and has to be copied to the SD card. Its format should be self explanatory.

MilesLives, 2024-03-11:
Are more Patches (i.e. Sounds) available? Thanks a lot, see you soon. M

ellisgl, 2025-02-23:
From what @laktosterror said "_A third wave osc would be great, for even more versatility. A suggestion there would be to have two tabs above the filter-column in the ui, for vco 1 and 2_" that would be awesome.

Also from looking that the plain text of the patches - I'm not sure what what is for the most part.