rsta2/minisynth #13: New Features

michelemaroni, 2021-04-07:
This is a really interesting and powerful project, that can go well beyond the origina intent of showcasing the bare metal environment for raspberry pi.

It could actually be the starting point for a bigger community (like [zynthian](https://zynthian.org), [ELK audio os](https://elk.audio/audio-os/), and [samplerbox](https://www.samplerbox.org)) and have also a market.

I would propose a few new features that would really increase the maturity of this project as a standalone synth:
-an option to set the duty cicle of the square wave in the VCO section
-an option to select PWM wave and an option to select PWM speed in the VCO
-an option to set the semitones of the second VCO in the range of +/- 24
- the keyboard tracking feature, that would slightly increase/decrease the cutoff frequency of the filter as higher or lower notes are played.
- support for hifiberry or generic pcm5102/5122 DACs through gpio
- Monophonic mode like glissando (note slowly change pitch when pressing another key)/legato(note immediately change pitch when pressing a new key)/staccato (like legato but filter is retriggered with the new note)

It would be interesint to add description of filter used (number of poles, pass modality) by the synth in the description. It is possible to bypass the envelope generatore in the filter section?

I wish i could actually contribute to the proposals but unfortunately i don't know c/c++.
I would be happy to support the project in any mean possible.

Thanks for making this really exciting project available.

rsta2, 2021-04-07:
Thanks for being interested in MiniSynth Pi, for your suggestion for new features and for your offer to support the project! I will see, what I can do. Unfortunately I have not enough time to serve all my projects in short terms and I'm jumping around between them a little bit. For instance I'm also writing a documentation for Circle at the moment. MiniSynth Pi is based on it.

Because I'm not a professional audio programmer, I have to search around on the Internet often to learn about new features. So I will do for the features, you have requested, but it will take some time.

The monophonic mode with glissando has already been requested in issue #12, but glissando is not that easy to implement in MiniSynth. There is already I2S support for the PCM5102A in MiniSynth. I will think about the other features.

The filter in MiniSynth Pi was implemented based on [this document](https://webaudio.github.io/Audio-EQ-Cookbook/Audio-EQ-Cookbook.txt). It's the low-pass filter (LPF). The envelope generator cannot be bypassed yet, but the parameters can be set in a way, that it has no effect on it (attack 0ms, decay 10000ms, sustain 100%, release 5000ms).

rsta2, 2021-04-11:
There is support for the PCM5122 DAC now in Circle. The PCM5102A was already supported. This will allow to support the PCM5122 also in MiniSynth Pi, but I need to wait until it's clear, what happens with the HDMI audio support.

michelemaroni, 2021-04-11:
Thank you very much for the quick implementation.

I also noticed the new entries in the project section.

With reference to portamento/glissando, legato and staccato, I found this nice [forum posts](https://goldmidi.com/community/resources/legato-and-portamento-for-synthesizers-explained.37/) that provide some explanation.
I'll try also to summarize:

In glissando there is a transition between the current note played and the new note played, the rate/speed at which this transition occurs is determined by the portamento, which basically sets a time constant. The time at which the transition occurs is proportional to the frequency distance of the two notes. Thus if the portamento is very low you basically get legato, if it is high the greater the distance between two notes the slower is the transition. So this is difficult to implement as it would require to integrate the portamento constant and the priority between keys played in the oscillator section. 

Another thing to be aware of is that the last note pressed is the one setting the frequency of the oscillator: When a first note is pressed but not released, and then a second one is pressed and then released the frequency should go from the first note to the second and then back to the first.

Staccato might probably be the easier thing to start with and would provide the foundation for implementing legato and glissando later on. The synth is already polyphonic and each time a new note is pressed a vco is activate with the filter and the envelope settings. Thus legato could be implemented as a modification of the polyphony handling: If a first key is pressed a note is generated, when a second key is pressed the previous note is muted (with a cc note off signal maybe?) and a new one is generated. If the previous key is still pressed when the the second key is released the initial note is retriggered.

I hope this provide some help on the way forward. 
I would be happy to provide further information on this and any other issue related to this project.

rsta2, 2021-04-12:
Thank you very much for these explanations and the link to the forum article. This discussion is very helpful for me and I'm willing to consult you further to go forward. Thanks for your offer.

How I understand it, the current mode in MiniSynth Pi is polyphonic staccato. So the modifications for monophonic mode with legato and alternatively glissando (aka portamento/glide) may be:

1. The number of active voices has to be reduced to one. Polyphonic portamento (see the forum post) seems to be too much for this project currently.

2. For Legato, when playing note to note, do not re-start the EGs with the attack phase. Re-start with decay phase at 100% level instead? (When not playing note to note, with monophonic mode the previous note would be muted already, because there is only one voice.)

3. For glissando regularly slide in pitch with each new note. The time for frequency transition is proportional to the frequency distance of the two notes. The glide rate is set by a parameter ("portamento time"). What is a usual value for this parameter, when it is at 100% and the slide takes place between two semitones (e.g. C -> C#). Is this parameter linear or is there some curve to be applied? The EGs are re-triggered once, when a new note is pressed?

> Another thing to be aware of is that the last note pressed is the one setting the frequency of the oscillator: When a first note is pressed but not released, and then a second one is pressed and then released the frequency should go from the first note to the second and then back to the first.

For which mode does this apply (legato or glissando)? What happens for three pressed notes, for example (do I need a note stack?):

	note on note C -> playing C
	note on note D -> playing D
	note on note E -> playing E
	note off note E -> playing D
	note off note D -> playing C (?)

or

	note on note C -> playing C
	note on note D -> playing D
	note on note E -> playing E
	note off note D -> no change
	note off note E -> playing C (?)

rsta2, 2021-04-16:
PCM5122 DAC support is now in the MIniSynth Pi repo. It has been tested with a PiFi DAC+ v2.0 hardware.

michelemaroni, 2021-04-17:
Hello, Thanks for implementing the support to pcm5122. That's great

> 1. The number of active voices has to be reduced to one. Polyphonic portamento (see the forum post) seems to be too much for this project currently.

Indeed I'd rule this out. The forum says it is possible, but 've ever really seen it on a synth.

> 2. For Legato, when playing note to note, do not re-start the EGs with the attack phase. Re-start with decay phase at 100% level instead? (When not playing note to note, with monophonic mode the previous note would be muted already, because there is only one voice.)

Could actually be the way to go, [i found a similar example on youtube](https://www.youtube.com/watch?v=hIcyO6kLy1I) 

> 3. For glissando regularly slide in pitch with each new note. The time for frequency transition is proportional to the frequency distance of the two notes. The glide rate is set by a parameter [...]

Transition between notes can last up until a[ few seconds](https://www.youtube.com/watch?v=S6M843hOkQs).
I found one example of portamento javascript code [here](https://github.com/Tonejs/Tone.js/blob/c313bc6/Tone/instrument/Monophonic.ts#L26), and [here](https://synthhacker.blogspot.com/2013/04/portamento-voice-spreading-from-monopoly.html) i found some informal description of the portamento equation. Unfortunately no values provided, however it is insightful.

> For which mode does this apply (legato or glissando)?

Both

>What happens for three pressed notes, for example (do I need a note stack?):
> 
> ```
> note on note C -> playing C
> note on note D -> playing D
> note on note E -> playing E
> note off note E -> playing D
> note off note D -> playing C

This one is what happens, indeed it is supposed to make the synth sounds like [hammer on and pull on a guitar](https://www.youtube.com/watch?v=B6vJDprnRfo) (hope this helps).

rsta2, 2021-04-20:
Thanks for the response and the interesting links. I started to work on the glissando / portamento.

michelemaroni, 2021-04-20:
Would it be possible to have a new release with the last changes in the meanwhile? I tried to build it using WSL2 but I am having some issues as it says that the prefix for the toolchain is not valid.

Thanks

rsta2, 2021-04-20:
The problem is, that there is an important issue on the RPi 1 at the moment. It freezes after a while there. I haven't found the reason yet. If you have RPi 2-4 I would build a version for you, but you would have to tell me, which RPi model you have and which audio interface you want to use (I2S or the normal headphone jack).

To build MiniSynth Pi you need a toolchain, which can be downloaded [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads). You have to specify the path and prefix of the toolchain commands as second parameter of the configure script.

michelemaroni, 2021-04-20:
I have a raspberry pi 3, and a pifi dac+ 2.0.

I'll definitely need to look deeper on how to build the image, since I installed the toolchain using apt, so maybe some dependencies are missing.

rsta2, 2021-04-20:
Here is a build of the current MiniSynth Pi sources for the RPi 3 with I2S. Please see the file INSTALL.md in the archive.

[minisynth-rpi3-i2s.zip](https://github.com/rsta2/minisynth/files/6345616/minisynth-rpi3-i2s.zip)

michelemaroni, 2021-04-22:
Hello, I tested the build and it works nicely.
I have some different suggestion/changes about the GUI and the VCOs tuning, but i'll later open a different issue for that.

Thanks for the image.

rsta2, 2021-04-22:
You are welcome. Good to know, that it works for you with the PCM5122 DAC. I already have some glide implemented, but probably not practically playable yet. Unfortunately I was facing problems and I'm trying to find the reason for issue #6 now. I also have other parallel projects, so it will go a bit slowly.

rsta2, 2021-04-23:
I've solved some issues in MiniSynth Pi, including issue #6. I also pushed the initial attempt to add monophonic mode with portamento to the branch *monophonic-mode*. The glide is linear at the moment and takes 10-1000ms per semitone. Perhaps we need an other function and other parameter range here.

It would be great, if you can give me feedback. You will find a new kernel image attached here. You simply need to replace the file *kernel7.img* from the last complete archive with the new extracted file on the SD card. There are two new parameters in the GUI: Voice Mode and Portamento Time.

[kernel7.zip](https://github.com/rsta2/minisynth/files/6365985/kernel7.zip)

michelemaroni, 2021-04-23:
Hi, made a test and both the features seem to work.
There is something strange with the filter and volume as it is not consistent between different notes, but i'd do some more test since I am not sure is set the filter correctly and I used a digital piano with weighted keys.
However, this was an impressive job, and I think we can close this issue.

I'll open two more, one hardware related and the other GUI related, but I know this is a side project and I don't expect a fast reaction. I'll propose this guy[Floyd Steinberg](https://www.youtube.com/watch?v=Aq9Bavd7OqY) to make a video about this project, with the hope that it gets more attention, and hopefully find some people willing to contribute in the development. Would you be happy with this proposal?

closed 2021-04-23 by michelemaroni

rsta2, 2021-04-24:
Hi, thanks for the positive feedback. Yes, there may be still a problem with the VCF and VCA. On a note off event, when the next note to be played comes from the note stack, I didn't knew, if the EG of the filter and/or amplifier have to be re-triggered and if so, with which velocity? At the moment the EGs are only triggered, when a new note on event is received.

I will schedule this project from time to time to implement new features. It would be great to have contributions for MiniSynth Pi. Especially when it comes to things, where you need to be familiar with synth programming, such support would be great. Even that I'm not sure, if this project is already good enough to talk about it in such a channel, more attention would be good for the project, of course.