rsta2/minisynth #14: Enable setting of GUI resolution, enable user input of precise values for GUI settings, add hardware support for USB and GPIO touchscreen

michelemaroni, 2021-04-27:
[image]

Hello,

Thanks again for the experimental build.
I made some tests and I've come up with some new potential improvements.

1) Enable the user to set the resolution for the GUI:
 Editing some configuration file before powering on would be sufficient.

2) Enable user input of precise values for GUI settings:
Basically sometimes the steps changes provided by arrows are too big. If a user could directly enter the desired value (tapping twice on the value or somethin like that) it would be very useful. For the moment this is possible by manually editing the patches correct? 

I'd also change the unit and the values for some sections, but I need more time to carefully check, note them down and report to you. E.g. the tuning of oscillators would be better set using the cents as a unit, and probably adding also a key transpose feature for each available oscillator. This could be achieved by multiplying the oscillator base frequency for a contants.

For key transposition, one oscillator (O1) is set to a frequency and the second one (O2) is set to a frequency that corresponds to a specific interval in semitones:
O2  = Nr_Semitones*[12√2](https://en.wikipedia.org/wiki/12_equal_temperament)*O1

For tuning by cents it would be:
O2=Nr_Cents*[1200√2 ](https://en.wikipedia.org/wiki/Cent_(music))*O1

Then for two active oscillator I would have:
O2 = ΔFreq_of_Cents + ΔFreq_of_Semitones +O1

Let me know if this is confusing, I'd try to explain it better.

3) Add support for resistive and capacitative touchsreen via gpio or usb: 
I understand that hardware support for any user device would be way beyond you interest and resources., however the majority of users buy chinese display that are all based on the [waveshare chipsets](https://www.waveshare.com/product/raspberry-pi/displays/lcd-oled.htm?___SID=U&dir=asc&order=price). The [most common](https://discourse.zynthian.org/search?q=waveshare) are display of 3.5-5 (generally resistive touch via GPIO) and 7 inches (generally capacitative touch via usb) displays. How difficult and time consuming would be to add support for those?

rsta2, 2021-04-29:
Thanks for the new suggestions for improvements.

With the GUI you hit a difficult point. It is kept simple by intention, all parameters use the same up/down buttons and the positions of the different controls are fixed and do work with the 800x480 screen resolution only, which work with most displays and the official 7" Raspberry Pi touchscreen, which is already supported by the Circle framework and by MiniSynth Pi. To support the 3.5" Waveshare display would require a totally different arrangement of the controls and more tabs, because the display resolution is much smaller.

For the moment I would focus on the supporting the 5" or the 7" Waveshare display. I would buy one of these displays to be able to test it, but not both. So it would be good to know, which display is used by more synth users? The support has to added to Circle first. I would stay with the 800x480 resolution at first. Is there a datasheet for the GPIO or USB touchscreen function somewhere?

Editing the parameters directly could be probably implemented, by using an additional popup window. The user taps on the parameter value and the popup opens, where the new value can be entered. I will have to open a screen keyboard for this purpose.

The modifications for the oscillator section should be possible to make. I will need another two fields for the cents detune and the key transpose in the GUI.

michelemaroni, 2021-05-04:
Hi,

I couldn't find a datasheet for the display, however there is an official repo with the [drivers for the different displays](https://github.com/waveshare/LCD-show) and a [wiki](https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)).
I'd guess that the majority of the user go for the 5'in display, but it comes in two different flavours: the one with resistive touch uses and the one with capacitative touch.

Then I'd drop the plan for the 3.5 display then, since not only is difficult, but also those displays are very limited: I had one and gave it away, and upgraded to a 7 inch. Instead scaling up the resolution to 1024×600 would it be feasible?

rsta2, 2021-05-06:
Thanks for the info about the different displays and the links. I would suggest the following: I will buy a 5 inch HDMI display (probably LCD (B)) with USB touch function. You have the 7 inch LCD (C). So we can test MiniSynth Pi with both display sizes and I have to implement the USB touch interface only, at least for now. It should be possible to provide two (fixed) screen sizes (800x480 and 1024x600) in a way, that on the larger display the different controls get more room.

I guess, I have to place the different parameters on the MAIN tab in a different way in the GUI at first, because there is not enough room for new parameters for the oscillator any more. Maybe there must be another new tab (e.g. one "VCO / VCF" tab, instead of MAIN, and one "VCA / FX" tab)? On the other hand, I think, it's good to have everything "in sight", without the need to change the tabs all the time.

What do you think?

michelemaroni, 2021-05-06:
The idea of splitting the GUI by section sounds actually good, it is for example the same strategy used in high end synth and workstations like [the kronos](https://youtu.be/Vdb5CYxpx0c?t=239). The important thing is having a meaningful grouping of parameter within the same window.

I'd propose to put VCO and LFO in one tab, the EG in another and then one for the FXs where we could foresee slots for different fx to be chained together.

About the display, yes i have the one with usb , and sounds like a good strategy.

rsta2, 2021-05-07:
So I think, we should go this (display) way. I've ordered a 5 inch Waveshare LCD B with HDMI and USB.

I think, for a start I will add one new tab to the GUI to get more room for the VCO. The parameters have to be re-ordered any way later.

Please note, that this is not a high performance audio design. With 24 parallel voices I have already about 60% of the possible CPU time used to calculate the audio samples on RPi 2-4. It's already more on the RPi 1 with 4 voices. There is one point for improvement on the RPi 2-4: It's currently running at 600 MHz only, while for instance the RPi 3B supports 1200 MHz. But you need a fan to continuously use this and not anybody has one.

rsta2, 2021-05-12:
Thanks for explaining this! I think, the number of polyphonic voices can be made configurable, as suggested. I got the 5" Waveshare touchscreen. Generally it is working, but not with touch yet, of course. I'm currently preparing a new Circle release and will come back next week.

rsta2, 2021-05-15:
Attached there is a first test version of MiniSynth Pi, which works with the Waveshare 5 inch LCD (B) touch screen and should work with the 7 inch LCD (C) too. I'm not totally sure, if it works with the 7 inch version and want to test this here. If it does not work, we have to step back and test with an other program.

Please don't wonder about the unusual GUI. I only modified the program, that it auto-detects the screen size, but effectively it still only uses 800x480 pixels. This test is specially for the touch function, not for the GUI itself. You only have to replace the file *kernel7.img* in the archive, I did send you earlier. The 5 inch version needs some calibration info, but I guess, the 7 inch version works without it.

[kernel7.zip](https://github.com/rsta2/minisynth/files/6485994/kernel7.zip)

michelemaroni, 2021-05-16:
Unfortunately it does not work.

rsta2, 2021-05-17:
OK, that was possible, unfortunately I was too quick. Thanks for testing. Then we have to go a step back and need some information from your touchscreen (vendor and device ID, HID descriptor). Attached there is another kernel image, which writes a file *circle.log* on the SD card. Can you please run it, with the touchscreen connected. You only need to replace the file *kernel7.img* with the one from the extracted .zip archive. Please run it once and copy the file *circle.log* from the SD card and post it here. The touchscreen will not work yet. Thank you!

[kernel7.zip](https://github.com/rsta2/minisynth/files/6492617/kernel7.zip)

michelemaroni, 2021-05-21:
Hello, sorry for the late reply, had quite a busy week.

Thanks for the image, I run the test and got you the log file that you asked for.

Please find it attached below.
[circle.log](https://github.com/rsta2/minisynth/files/6523588/circle.log)

Please let me know if there is anything else that I can help you with in the meantime.

Best regards

rsta2, 2021-05-22:
No problem and thanks for testing! I was able to retrieve the needed information from the log file.

The previous test did not work, because your touchscreen has a different USB vendor and device IDs from my own touchscreen. After some reading on the Internet, I expected them to use the same USB chip and vendor/device IDs, but they don't. Also the format of the HID touchscreen reports is different.

I think, I should try to find a generic solution for all USB HID compatible touchscreens here, instead of a specific driver for our two screens, but this requires additional work. I hope, I can complete this until next weekend and will post a new test image then.

rsta2, 2021-05-27:
I have implemented a generic USB touchscreen driver. It works on my Waveshare 5 inch LCD (B) and hopefully on all USB touchscreens, but I expect quirks with some touchscreens, so there is some uncertainty. It would be great, if you could test this driver with the attached test kernel image on your own touchscreen.

The test program initializes the touchscreen and waits for touch events. When a finger is pressed on the display, a `Finger #N down at X / Y` message should be displayed on the screen, where N is the finger ID (1..10) and X / Y is the pixel position of the finger. Other messages are `Finger #N moved to X / Y` and `Finger #N up`. According to the HID report descriptor, your screen may support up to six fingers pressed at once.

If it generally works, please have a look at the X / Y coordinates. The first value should range between 0 and 1023 for your screen and the second between 0 and 599. If this is not the case, your screen needs some calibration for this driver. I would tell you next, how this works.

[kernel7.zip](https://github.com/rsta2/minisynth/files/6552730/kernel7.zip)

michelemaroni, 2021-06-03:
Hi sorry for the delayed answer.

I made the test with the image you sent me and the driver worked very nicely: touche seemed accurate and very responsive.

The values were a bit different for my screen but i guess is due to the pixel size.

Thank you.

Best,

MM

rsta2, 2021-06-04:
Hi, good to know, that it's working now. Thanks for testing! There will be a calibration tool available, which can be used, if the touch coordinates do not exactly match the screen coordinates. I will work towards using the touchscreen driver in MiniSynth Pi with other screen sizes than 800x480 now.

rsta2, 2021-06-06:
The GUI is now variable for different screen sizes. The design is not perfect, but should work. Also the USB touchscreen should work. I implemented a calibration tool for the USB touchscreens.

Now I will implement the support to input precise parameter values using a screen keyboard and maybe another tab for the envelope generators to get room for more parameters for the VCO until next weekend. I will send another test kernel image then.

michelemaroni, 2021-06-25:
Hi,

I made a test with the new images and the touchscreen works.
The input of values has still some issues:
Tapping on a filed correctly opens the screen keyboard, however the values get round/approximated to the step increase allowed (e.g. if i set the lfo rate to 33% it goes to 34%).
I should spend some time and come back with a proper list of values and change rate that each parameter should have.

One thing that i noticed and i like quit a bit is that there is a velocity control of the VFC cutoff.
A dedicated control for it would be a nice to have.,
Also, what about a midi learn in the pop up keybord? You touch a filed, you get the keybord for inserting the parameter, and in the keyboard also a button that activates the midi learn function.
Midi learn is a feature that listens for midi messages and assign a parameter to the received midi input.
For example, I tap on the VCF cutoff and i tap on midi learn, than if I turn a knob on my master keyboard, the program assign the control of the cutoff to that knob by its midi CC.

At this point I'd suggest to finish the implementation of the monophonic mode, as you already made some interesting achivements in there.

What about also having a new release with the latest developments?

Best

rsta2, 2021-06-28:
Hi, thanks for testing the new kernel images. Good to know, that the touchscreen works. I fixed the issue, that entered parameter values were adjusted to the step increase setting. Now this is only done, when the down/up buttons are pressed.

Yes, the VCF cutoff frequency is influenced by note-on velocity at the moment. Can this be generally disabled or on user choice only?

The MIDI learn function is a great idea, but I think this is an advanced feature and this is not that easy to implement. The numeric screen keyboard is a feature of the underlying GUI library and unfortunately another button cannot be added there. I added this feature as an item to the GUI project for later.

I will prepare a release now with the new developments, because there hasn't been a release for over a year now. Then we can continue with the monophonic mode.

A list of the value range and change rate, that each parameter should have, would be very useful.

michelemaroni, 2021-06-28:
Hi,

The control of VCF cutoff by velocity is in general a desirable feature.
However it would be useful to make clear it is engaged and to control how much and in which direction the velocity modulate the VFC cutoff. 
This feature is generally coupled also with the keyboard tracking, where the cutoff frequency can be adjusted to a certain percent and in both directions according to the  pitch of the notes:
e.g. i set up my synth with a certain vcf cutoff frequency, and it sounds good at C4 octave but too open in C3 and C2 octaves and too closed in C5 and C6 octaves. With keyboard tracking this gets adjusted.

If it is not clear have a look at page 10 of the [polysix manual](https://www.korg.com/us/support/download/manual/1/194/1640/)

rsta2, 2021-06-29:
Ok, thanks for explaining this!

rsta2, 2021-07-02:
@michelemaroni Release v4.0 is out. Thank you for your support for this release!

rsta2, 2021-07-20:
Hi @michelemaroni,

while trying to develop MiniSynth Pi further I was facing a serious problem. I'm a system programmer, not a musician or professional audio programmer and playing keyboard is something, I'm not able to do well.

Now that I tried to improve the glissando / portamento feature, I have the feeling, that it sounds strange, but I do not have the experience in this field, so that I can judge this reliable and so I cannot improve it. Perhaps with your help it is possible to answer a number of questions, but in the end it remains a difficult process.

I guess a synth, which can compare with other established professional or open source synths, needs much more knowledge than how a simple oscillator or filter can be implemented. An instrument, which really sounds well and can be used to make music, contains many specific characteristic curves for example, which need a lot of work to design them.

It is possible to implement professional synths based on the Circle framework, which is also the foundation for MiniSynth Pi. But I am not the person, who can do this. Unfortunately I think, it will not work out in the end. Thus I think, MiniSynth Pi should remain what it is, an example, which shows, that implementing a polyphonic synth based on Circle is possible.

Best regards,

rsta2

michelemaroni, 2021-07-26:
Hello,

Yes I understand. I think that is not a big deal the keyboard playing, and also the proposal here is not to implement a professional synth but rather further improve an example that turned out to be already nice.
This on one side might attract some hobbyist willing to contribute to minisynth development and as a side effect to circle.

However it is really you call.

In the meanwhile I made a fork and I will enjoy the further developments that happened in the last months.
I'll remain available in case you want to resume it in the future.

Thanks and good luck.

rsta2, 2021-07-28:
I will keep this in mind. Thanks.

pinkmouse101, 2022-08-23:
Hi folks
 Can you supply the model of the 7" WS screen you are using please. I have this one:

https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)

and it's not displaying correctly, (white screen with what looks like rolling noise bars)

Cheers
Al

rsta2, 2022-08-24:
I think @michelemaroni tested it successfully with the same display. Are you sure, that you have setup the display correctly? According to the description of the display you have to add the following options to the file *config.txt* on the SD card:

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

pinkmouse101, 2022-08-24:
Ha, me bad!

I cut and pasted the config, rather than typing it in, which worked fine. Now display working, but no touch. To use the calibration tool do I copy all three of the Kernel files to the main directory, (after renaming the originals), the ReadMe isn't that clear?

Edit: I copied all three files and got the message:

00:00:61.6 kernel: Touchscreen not found 

...In nice red letters!

I'll try plugging it into my PC and see if it works there.

cheers
al

rsta2, 2022-08-24:
If you get that message from the calibration tool, the tool should be installed all right. Can you please try the calibration tool again after adding the following option to the file *cmdline.txt* on the SD card:

```
usbpowerdelay=3000
```

This option must be added to the first line of *cmdline.txt* and delimited by a space, if there are already other options in this file. This gives the touchscreen more time (3000 ms) to settle after powering on the USB port.

pinkmouse101, 2022-08-24:
Thanks Rene, but no joy. I'm suspecting the problem is on the display itself, not with minisynth. I'm away this weekend, so won't be playing with it, but I'll do a bit more testing next week to see if I can sort out the problem. It was working fine, I tested it with a bare Raspbian install when I first got it about 18 months ago, (I was going to use it as a CNC controller, but went in another direction for that), so it's been in storage for a while. I suspect the micro USB connector. I hate them with a passion!
cheersal

rsta2, 2022-08-24:
OK, I understand.

pinkmouse101, 2022-09-03:
Well, here we are again...:-)

Port cleaned and tested, new cable acquired, and it gets slightly further than last time, it sees the waveshare, but than bottles out. I've attached a screen pic, I can't seem to find a log file...

[image]

Hope this is of use, and thanks again for your help

Cheers
al

rsta2, 2022-09-04:
Thanks for sending the log screen. For some reason the HID report descriptor cannot be fetched from the touchscreen, which is required for operation. I guess there is a problem in Circle regarding the HID report descriptor handling. I have to analyze this, which will take some time.

pinkmouse101, 2022-09-04:
Thanks Rene, no rush. I have a Minidexed and MT32_pi to build as well! :-)
Just thinking how I can help - could I lend the display for you to test? Or do you have a paypal or something so I can send you a tenner for coffee?
cheersal

rsta2, 2022-09-04:
Perhaps I have already found the problem. Can you please extract the file *kernel7l.img* from the attached .zip file, replace the same file on the SD card with this version and try again. This is a modified version of the touchscreen calibrator, which sets the USB device configuration, before reading the HID report descriptor. A donation is not necessary. It's already a great help, if you return the test results. Thank you.

[kernel7l.zip](https://github.com/rsta2/minisynth/files/9484542/kernel7l.zip)

pinkmouse101, 2022-09-06:
That works!  And my touchscreen doesn't need calibration! :-)

However, it's still not recognized in the main program, I assume you need to change that as well?

cheers
al

rsta2, 2022-09-07:
Great! Thanks for testing. Yes, the main program must be updated separately. You find an compressed image *kernel7l.img* of MiniSynth Pi in the attached .zip file. This version has been built from the master branch with the current Circle version on the *develop* branch.

Please note that it has some additional modifications, compared to MiniSynth Pi v4.1. By default it uses the headphone jack output, which is available on the Raspberry Pi 4B board. If you want to use I2S you have to add the option `sounddev=sndi2s` to the file *cmdline.txt* on the SD card.

[kernel7l.zip](https://github.com/rsta2/minisynth/files/9505299/kernel7l.zip)

pinkmouse101, 2022-09-08:
Initial testing good, all seems to be working well! Now to build a case and sort out midi and DAC.

Thanks for all your work and help Rene. :-)

al

rsta2, 2022-09-09:
You are welcome. Thanks for testing. Have fun!

pinkmouse101, 2022-10-11:
That works!  And my touchscreen doesn't need calibration! :-)
However, it's still not recognized in the main program, I assume you need to change that as well?
cheersal