rsta2/minisynth #15: 5" touchscreen display does not work

jknutson, 2021-06-01:
Hello,

I have the 5" adafruit touchscreen display, and while it works well for other applications (assuming the correct 800x480 resolution is configured), it is not working for minisynth.

The behavior I am experiencing is the same as if the resolution is not set correctly, despite it being hardcoded in minisynth to the correct 800x480. I did also update the `circle` submodule to pull in updates regarding the cmdline.txt resolution fix, but that did not work either.

The application works fine when on a display that can scale the resolution. Let me know if there are any ideas or other information that I can provide, I would love to get this working with my little screen!

Thanks

rsta2, 2021-06-01:
There is currently support for the Official 7" RPi touchscreen only in the MiniSynth Pi sources on GitHub. I'm working on a generic driver for touchscreens with USB HID-class standard interface, but there is only a test program in issue #14 available, which awaits testing. If your display has an USB interface, there is some chance, that it may work with MiniSynth Pi in some weeks, but I'm still working on this. I will let you know, when a test version is available, so that you can test it. Can you tell me, which of [these displays](https://www.adafruit.com/category/338) you have?

jknutson, 2021-06-01:
Thanks for the information!

I believe I have the first display in the search you posted [(this one)](https://www.adafruit.com/product/2260), which I thought was functionally the same as the 7" display but there must be some differences.

I will try out that test program when I have some time in the next day or two on my 5" display.

Thanks for the work you are doing on minisynth and circle, very cool applications!

rsta2, 2021-06-01:
Thanks for appreciating MiniSynth Pi and Circle! Unfortunately your touchscreen will probably not work with the test program. According to the description at the Adafruit site, your touchscreen emulates a mouse and does not work in digitizer mode, which is supported by the Circle driver, which is in development. Furthermore the mouse mode, your screen is likely to support, is different from the normal mouse mode with differential coordinates, which is implemented in the Circle USB mouse driver. The Adafruit screen sends absolute coordinates. I have a problem here, because the Circle mouse driver was not developed by myself in the current version, so I cannot easily change this. Perhaps this can be implemented later, but a quick solution is unfortunately not in sight.

jknutson, 2021-06-02:
That makes sense, thanks for digging in!

closed 2021-06-02 by jknutson