rsta2/minisynth #18: mini synth don't start

jojo-monk, 2021-11-15:
i am trying to install minisynth on a raspberry pi 3 B +, i copied all the files to the sd card, but the raspberry does not want to start, nothing is displayed on the screen ...

rsta2, 2021-11-16:
I tried v4.0 again from scratch on a RPi 3B+ and it worked.

I need more info. Did you use v4.0 and installed all files from the .zip archive on an empty SD card with a FAT partition (first partition on the SD card)? Which (USB) devices do you have connected? Which display do you use? What happens on the screen, is it blank black or is a rainbow image displayed?

Can you please try to start without any USB device connected?

jojo-monk, 2021-11-16:
Thanks,
that's it, mini synth starts up, it's the mac keyboard that blocks the startup.
By cons I can not get the slightest sound, I tried with a usb e-mu keyboard is not detected, I also tried with a homemade midi controller based on arduino leonardo, it does not not working either.
Is it possible to control minisynth by ssh, because suddenly I have no way to interact (my keyboard and my mouse are not detected).
My project is to make a very simple onboard synth with my arduino controller (without screen, keyboard or mouse), do you think that is feasible?

rsta2, 2021-11-16:
Can you please test the attached kernel image. It's the version for the headphone jack (not I2S). Just decompress it and copy the file *kernel7.img* to the SD card (replace the old image). You can also rename the file *config.txt* to something else and you will get messages from the program on the screen. The messages destroy the GUI, but that's not important here. Perhaps this helps to find the problem.

[kernel7.zip](https://github.com/rsta2/minisynth/files/7548583/kernel7.zip)

rsta2, 2021-11-16:
The file to rename to get log messages is *cmdline.txt*, not *config.txt*. My apologies.

jojo-monk, 2021-11-16:
without the mac keyboard, I now have sound, and the usb midi keyboard is well detected.
If I connect the apple keyboard, I have these messages showing:

[image]
When I boot up with the apple keyboard already connected, I have this
[image]
:

rsta2, 2021-11-16:
The problem may be the USB hub in the Apple keyboard, which has a 9-0-0 interface. That means it has only a single transaction translator (TT) for low- and full-speed devices. Such hubs are not well supported but Circle. USB hubs like the internal hub of the RPi 3B with interface 9-0-2 are better supported. They have multiple TTs. Can you please try this, without using the keyboard hub? Thanks for testing.

jojo-monk, 2021-11-16:
I had the same error without using keyboard hub :
[image]

rsta2, 2021-11-16:
The problem is, that the request to get the HID report descriptor fails. The failed assertion is a followup error. Can you please make another test with the attached kernel image? I added more log output. Thank you!

[kernel7.zip](https://github.com/rsta2/minisynth/files/7549983/kernel7.zip)

rsta2, 2021-11-16:
Thanks for testing! Unfortunately that looks all right, but the -1 tells, that the control message request to get the HID report descriptor fails in the USB low level code. This can have many reasons and sorting this out would probably take long. For the moment I probably have to blacklist this keyboard in the [Circle Wiki](https://github.com/rsta2/circle/wiki/Device-compatibility-list). I guess, you have alternative keyboards to use with MiniSynth Pi.

rsta2, 2021-11-17:
After thinking and analyzing this problem again I guess, I have found the problem for the failed assertion. The failed request for the HID report descriptor has to be accepted, but it should not crash the system. There is another test kernel image attached. It would be great, if you could test it again with the Apple keyboard.

[kernel7.zip](https://github.com/rsta2/minisynth/files/7553311/kernel7.zip)

rsta2, 2021-11-19:
@jojo-monk Did you already had a chance to test the new kernel image?

jojo-monk, 2021-11-23:
Oh yes excuse me, i test this this afternoon

jojo-monk, 2021-11-23:
how to copy img file to sd card?

jojo-monk, 2021-11-23:
invalid disk image when i try to copy the file

rsta2, 2021-11-24:
Please install the release v4.0 on a SD card, extract the file *kernel7.zip*, downloaded above, to *kernel7.img* and copy it to the SD card. This replaces the old kernel image. You can also rename the file *cmdline.txt* to something else to get the log messages.

jojo-monk, 2021-11-24:
oh sorry i'm tired i thought it was a system disk image

jojo-monk, 2021-11-24:
now mini-synth starts up fine, even with apple keyboard plugged in, but keyboard not working

[image]

rsta2, 2021-11-24:
Thanks for testing and no problem, I'm happy that you help me to sort this out.

This behavior (no crash, but keyboard does not work) was one of the expected. Of course a crash should not happen. Now is the question, why does the keyboard not work. In MiniSynth/Circle normal USB PC keyboard are used with the `int3-1-1` interface, which seems to initialize all right with your keyboard.

The `int3-0-0` interface is meant to be used for keyboards with a special layout. I suppose, it is not needed here. There is some chance, that the failure when reading the HID report descriptor stalls the whole keyboard. I tested an other keyboard here and this was not the case there. It still worked, even when reading the HID report descriptor on the `int3-0-0` interface failed. With MiniSynth you can only check this, if you have a headphone attached to the 3.5" headphone jack and press a key, which is mapped to a MIDI note (e.g. the "C" key). Does it generate some sound then?

Perhaps you have checked this already, but if not, please do. I will add an `usbignore=` option for the file *cmdline.txt* then, with which one can prevent using the `int3-0-0` interface. Hopefully your keyboard will word then. Thanks.

jojo-monk, 2021-11-24:
No matter which key I press, there is no sound.

jojo-monk, 2021-11-24:
With apple keyboard of course

rsta2, 2021-11-24:
OK, I will implement the `usbignore=` option. Thanks for testing.

rsta2, 2021-11-25:
So let's go for another test:

1. Please extract the *kernel7.img* from the following archive, copy it to the SD card, where MiniSynth Pi v4.0 is installed and run a test, if a sound is generated, when some key is pressed on your keyboard. This version tries to read the HID report descriptor also on the int3-1-1 interface. This may help some keyboard to work.

[kernel7-readhiddesc.zip](https://github.com/rsta2/minisynth/files/7601393/kernel7-readhiddesc.zip)

2. If 1. does not work, copy the following file to the SD card and test again. With this the USB interface int3-0-0 is additionally ignored.

[cmdline.txt](https://github.com/rsta2/minisynth/files/7601396/cmdline.txt)

3. If 2. does not work, extract the following archive and copy the extracted *kernel7.img* file to the SD card. The *cmdline.txt* file remains on the SD card. Run the test again. This version ignores the USB interface int3-0-0, but does not read the HID report descriptor on the int3-1-1 interface.

[kernel7-usbignoreonly.zip](https://github.com/rsta2/minisynth/files/7601400/kernel7-usbignoreonly.zip)

Fingers crossed. ;) Thank you.

jojo-monk, 2021-11-25:
I did the 3 tests: 1st test: failure, the keyboard does not work
2nd test: it works, we can make sound with the keyboard, but we cannot write.
3rd test: same result as the 2nd test

rsta2, 2021-11-25:
OK, so the *usbignore=* option patch with the entry in *cmdline.txt* helps, that you can generate sound. This is the only function of a PC keyboard in MiniSynth Pi as a replacement for a USB MIDI keyboard for test, if one does not have one. You cannot write with it and have to use the screen keyboard to input data. This is by intention.

Now I know everything to create a new MiniSynth release, which works with your keyboard too. I can do this not before the next release of the Circle framework, which is the foundation for MiniSynth, is out. In the meantime I think, you have a USB MIDI keyboard to use, which is better to play anyhow.

Thank you for all that testing again.

rsta2, 2021-12-03:
The Release v4.1 is out, which includes the bug fixes and describes the workaround for your keyboard in the *Troubleshooting* section in the *README.md* file.

closed 2021-12-10 by rsta2