rsta2/minisynth #19: Contribution I'd like to make

poltronSuperstar, 2022-04-07:
For the last few years, the majority of my code was written in high level languages; Python, Javacscript, and all my low level code was in Rust. I literally have not touched at C++ for a decade. All of that to say that my C++ is rusty as hell.

I have some time and I'd love to regain my past skill; this project is totally amazing, so contributing to it would be an honor.

I have a few questions:
* Are there features on the TODO list not to hard to get back to it gently ?
* What I'd really love to implement is a loop station
	* Is it ambitious ?
	* Would you have some tips, keywords to google, directions to points towards ?

Anyway, thanks again for this project :)

rsta2, 2022-04-08:
Thanks for appreciating MiniSynth Pi. Contributions are welcome. The TODO list is [here](https://github.com/rsta2/minisynth/projects).

Because MiniSynth acts as an example, how synths can be programmed using the [Circle](https://github.com/rsta2/circle) bare metal environment, it shouldn't become too big and complicated. That's why I think adding a loop station to it is out of scope for this project.

But of course you could create your own project fork, to do this. Unfortunately I cannot give much info about loop stations, because I do not have experiences with such devices. After some quick reading I guess, it's more a sampler, than an analog synth. Perhaps the Circle [sample/42-i2input](https://github.com/rsta2/circle/tree/master/sample/42-i2sinput) can help to implement it. The Circle I2S driver supports output and input of audio data.

rsta2, 2023-11-10:
There was no further reaction from the originator of this issue. I think, it can be closed.

closed 2023-11-10 by rsta2