rsta2/minisynth #20: Not a Problem, Just some pics!

pinkmouse101, 2022-09-28:
Build finally finished, complete with 3d printed case. Thanks for all who created and assisted! :-)

I'm loath to put the case up as printable files, as this is a very personalized build, but if Rene wants to host the Fusion 360 file, I'm more than willing to post it here, that way others can tweak as desired.

cheers all
Al
[image]
[image]

rsta2, 2022-09-29:
Wow, this looks great! If you want to contribute your construction CAD files, you can add another comment here with an archive file (e.g. .zip) included. Thanks!

VasilyMedved, 2022-09-29:
Wow. Just wow ))) (we talk about Owen Wilson grade "wow" here)
_In the next iteration I suggest to consider adding few rotary knobs for volume level, filter cuttoff, reverb amount levels_

pinkmouse101, 2022-09-29:
Thanks, I'll post the files at the weekend with some more info.

pinkmouse101, 2022-10-04:
Right, here we go. You will need the free, (or paid!), version of Fusion 360 from Autodesk to open this file. I'm deliberately not posting 3d printable files as my build used odd bits and pieces I had lying around, so any components you have will likely not fit this model, and I can't give links to buy particular parts I used, as many of them I forget where they came from! :-)

I don't have space to go into the Fusion editing process here, but there are many good tutorials out there, and I think I've done a reasonable job at naming sketches and layers to make things as simple as possible.

However, a few notes about stuff I did use. The touch panel I used was the waveshare 7" LCD C and after a few tribulations, it works well. I also used a USB B type connector, as it's a lot more robust than the Pi's onboard mini ones. That just runs to the 5v and GND pins on the header. The MIdi interface and DAC boards are not screwed in, I just used velcro, as during testing they went in and out a fair bit, however the Pi is mounted properly as it has a bit more weight to it.

You will need a selection of M3 screws, and I used heat set inserts for all the threaded holes, if you don't you'll need to reduce the size of the holes the screws thread into.

The design is sized to be printed on Ender 3 and up, go bigger and you'll need a bigger printer. ;-)

Well, enough waffle, here you go! Enjoy!

[image]

[minisynth case v4.f3d.zip](https://github.com/rsta2/minisynth/files/9710081/minisynth.case.v4.f3d.zip)

rsta2, 2022-10-04:
@pinkmouse101 Many thanks for providing the 3D printable files for your MiniSynth Pi case! I hope it will help MiniSynth Pi users to get such a great case, like it can be seen on your pictures.