rsta2/minisynth #22: V5.0 Works well!

pinkmouse101, 2022-12-04:
That's it really, after copying over the definition for my touchscreen, plugged in the card and works first time! Well done Rene, and thanks!

but.....( :-) ), any chance of having the midi input channel set onscreeen? I'm swapping stuff in and out at the moment, and it would be dead handy to be able to change  without opening it up and editing the configs each time. 

Cheers
Al

rsta2, 2022-12-04:
You are welcome and thanks for info! I will see, what I can do for MIDI channel selection using the GUI. Will take some time.

pinkmouse101, 2022-12-04:
Thanks, no rush 

al

rsta2, 2022-12-04:
Here is a first test version, which allows to edit the MIDI input channel in the GUI. Attached you will find a kernel image for the Raspberry Pi 4B in the .zip file. The MIDI channel can be set to 1-16 or *Omni Mode* (all channels, default). The setting will be saved with the currently active patch, when it is saved. Please let me know, if it works for you.

[kernel7l.zip](https://github.com/rsta2/minisynth/files/10149349/kernel7l.zip)

pinkmouse101, 2022-12-07:
Wow, what a response time! Well done! 

However.... the reason I had it open in the first place was to pull out the 4B, (for a Zynthian I have coming), and put in a 3B, borrowed from a 3d printer. I did check the new release on both though, and all was fine there, so that's one good thing.

I'll have a go with the new kernel anyway on the 3b this weekend, as I'm away with work 'till then.

Thanks again
al

rsta2, 2022-12-08:
Thanks. Here is a kernel image with MIDI channel selection in the GUI for the Raspberry Pi 2 and 3:

[kernel7.zip](https://github.com/rsta2/minisynth/files/10184063/kernel7.zip)

rsta2, 2022-12-16:
@pinkmouse101 I would like to know, if the added MIDI channel selection in the GUI is suitable for you. I would commit and push the updated source code to the repo then.

rsta2, 2022-12-21:
The modification for editing the input MIDI channel in the GUI has been committed and pushed to the repo.

rsta2, 2022-12-28:
The initial task for this issue is done, and I think, this can be closed.

closed 2022-12-28 by rsta2