rsta2/minisynth #25: Why only 24 voices on faster Pis?

Kirtai, 2023-03-06:
I'm curious why the Pi3 and Pi4 have the same polyphony as the Pi2.

Just simplicity or something else?

rsta2, 2023-03-07:
This is for simplicity. You can increase the number of voices per core [here](https://github.com/rsta2/minisynth/blob/master/src/config.h#L26). Because there are 4 cores you will get a total number of voices of 4 times this value.

By default MiniSynth Pi runs at only 600 MHz on the Raspberry Pi 2-4. You can add the option `fast=true` to the first line of the file *cmdline.txt* on the SD card to run it on the maximum supported speed. On the RPi 3 and 4 this may require a head sink and/or fan.

Kirtai, 2023-03-07:
Ah, thank you. 

Nice that it can be expanded if necessary :)

closed 2023-03-07 by Kirtai