rsta2/minisynth #26: 64 bit build fails on MacOS

u77345, 2023-06-06:
When building on MacOS the build silently fails for 64 bit build, when configured with:

`./configure 3 /usr/local/bin/aarch64-elf- 64`

make -d reveals an issue compiling lvgl:

```
    Pruning file `../circle/addon/lvgl/lvgl/src/misc/../hal/../lv_conf_internal.h'.
    Pruning file `/usr/local/Cellar/aarch64-elf-gcc/13.1.0/lib/gcc/aarch64-elf/13.1.0/include/stdbool.h'.
    Pruning file `../circle/addon/lvgl/lvgl/src/misc/lv_math.h'.
    Pruning file `../circle/addon/lvgl/lvgl/src/misc/lv_mem.h'.
    Pruning file `/usr/local/Cellar/aarch64-elf-gcc/13.1.0/lib/gcc/aarch64-elf/13.1.0/include/stddef.h'.
    Pruning file `string.h'.
   Finished prerequisites of target file `main.o'.
  Giving up on target file `main.o'.
 Finished prerequisites of target file `kernel8.img'.
Giving up on target file `kernel8.img'.
```

Workaround: 32 bit build creates viable kernel.img with:

rsta2, 2023-06-06:
Probably an incompatibility with your 64-bit toolchain. Have you tried an *aarch64-none-elf* target toolchain from [this website](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)? I'm using the 12.2.Rel1 version and MiniSynth Pi builds successfully on my Linux host with it. Unfortunately I cannot try this by myself on MacOS, because I do not have a Mac.

u77345, 2023-06-06:
Will test with that toolcain.

Another concern is the build failing silently on missing headers or errors in the headers, due to the -MG in Rues.mg. Also applies to the wire Circle project.
Is -MG absolutely necessary?

rsta2, 2023-06-06:
The option `-MG` is used for dependency checking. When you have problems with it, you can disable dependency checking by adding the following line to the file *circle/Config.mk* after `configure`:

```
CHECK_DEPS = 0
```

u77345, 2023-06-08:
The Linux version of the toolchain works correctly. From now I'm building it in Docker.

closed 2023-06-08 by u77345